rockchip-linux/meta-rockchip #1: 64 bit host - build fails for MPP

sat8, 2017-03-20:
Hi

My host PC is a 64 bit Ubuntu and somehow MPP fails to build Please find the error message below

openembedded-core/build/tmp-glibc/work/cortexa17hf-neon-vfpv4-oe-linux-gnueabi/mpp/git-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi/../../libexec/arm-oe-linux-gnueabi/gcc/arm-oe-linux-gnueabi/6.3.0/ld: codec/libmpp_codec.a(mpp_enc.cpp.o): relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
| codec/libmpp_codec.a: error adding symbols: Bad value
| collect2: error: ld returned 1 exit status
| mpp/CMakeFiles/mpp_shared.dir/build.make:214: recipe for target 'mpp/libmpp.so' failed
| make[2]: *** [mpp/libmpp.so] Error 1

the other day I was able to build the same module MPP on a 32 bit ubuntu host pc. So perhaps some compatibility issue that can be fixed promptly?

wzyy2, 2017-03-21:
There seems some environmental problems.
I would recommend using docker to solve environmental problems.

https://github.com/rockchip-linux/docker-rockchip

sat8, 2017-03-21:
Following the rockchip wiki for yocto solved all problems. Thnx and pls
close this issue

closed 2017-03-22 by wzyy2