rockchip-linux/meta-rockchip #2: meta-rockchip layer questions

sat8, 2017-03-20:
Hi 

This is more of observations and general question rather than burning issues in the layer.

I am trying to build a yocto image out of this and I have run into few interesting situations that I would like to highlight. I can create a separate issue for each of the below if it is preferred.

1. The gstreamer layer refers to some definitions in libdrm layer. However, openembedded-core brings in it's own libdrm layer. So I needed to get rid of the openembedded-core layer to be able to use the libdrm available in this repo. If this is not done, gstreamer build fails.

2. Even if gstreamer layer is built then gst-inpect will not pick up the plugins. It will blacklist them as gstreamer in fido or later is not 1.10.4. It's only when  the master branch of openembedded-core is used, gst-inspect-1.0 will see the plugins as legit. Perhaps the readme could highlight the exact version compatibility etc.,

3. I am not entirely sure on how the mali libraries in libgles need to be build and used in my rootfs. the reason being that bitbake complains about **'Nothing provides virtual/egl ....'** style errors if **rockchip-mali** is enabled in my conf. The only way I ever got to build libgles layer is to change the module name for graphics related modules from 'mesa' to 'rockchip-mali' in default-providers.inc.

4. (3) does not entirely work, though. I then need to extract libmali.so and copy into the rootfs and then create sym links. The rkvideo type plugins are then seen by gst-inspect otherwise they complain about undefined reference for gles2 etc.,

5. For using the hardware accelerated decoding and encoding, I have seen mpp deocde and encode plugins in gstreamer post running gst-inspect. Do I then require rkvideo plugins if I am not interested in displaying the video? 

6. Is gstreamer1.0-vaapi required for hardware accelerated video handing? 

7. libmali seems essential as this is where opencl is referenced. What is the proper way of building this module and including it my root fs? 

Thanks in advance
sateesh

wzyy2, 2017-03-21:
Hi,

You should include this meta : https://github.com/rockchip-linux/meta-rockchip-extra
guide: http://rockchip.wikidot.com/yocto-user-guide

Or you have to setup conf by yourself.

2. morty branch is only used in morty
5. No need, you can use QT or `gst-launch ! kmssink` to display the video.

sat8, 2017-03-21:
>    1. morty branch is only used in morty
>    2. No need, you can use QT or gst-launch ! kmssink to display the

Wow. The wiki link is the key. The image builds flawlessly now.

Thnx and pls close the issue.

closed 2017-03-22 by wzyy2