rockchip-linux/meta-rockchip #3: YOCTO SD card image not working

sat8, 2017-03-22:
Hi 

I have been able to build [SD card image](http://rockchip.wikidot.com/yocto-user-guide) for firefly-rk3288 but after the SD card is flashed with the image, the board does not want to boot. Also when the SD card is loaded in Gparted, it complains about GPT issue. One thing I have noticed is the boot.img generated in images folder is somewhat big (117MB) but I see in flash_tool.sh the BOOT_SIZE is set to 229376 (112MB). Could this be an issue?

[image]

Also, is it possible to generate a complete Android style system.img in yocto to jus
t cross check by burning the image onto emmc in windows?

Thanks

wzyy2, 2017-03-23:
Have you erase firmware in eMMC? 
Sd-card boot is the second device in the boot order.

The backup GPT table is usually at the end of the disk. To reduce image size, it was truncated, it's not an issue.

sat8, 2017-03-23:
>>Have you erase firmware in eMMC?

May be this is it :( . It doesnot want to boot at all even using emmc. Is the yocto image compatible with ASUS Tinkerboard (I know there is a machine conf in rockchip layer but just inquiring). I see that this board supports only SD card image? Can you confirm pls?

Would it then be a matter of taking the SD card image (not bootXXX.img in deploy/images)  and use flash_tool.sh and then plug into the SDcard slot?

Thnx

wzyy2, 2017-03-23:
SD-card boot is supported, but you shoud erase(zero) firmware in eMMC or hold maskrom button(in firefly, it might be different) to let rom load u-boot from SD-card. Tinker board don't have eMMC so rom always load u-boot from SD-card.

sat8, 2017-03-27:
Ok understood. I was able to get the image on timkerboard working . Thnx

closed 2017-03-27 by sat8