rockchip-linux/meta-rockchip #4: Wifi chip RTL8723BS driver for Tinkerboard

sat8, 2017-03-27:
Hi 

Tinkerboard comes with a rtl8723bs chip for wifi and Bluetooth. The TinkerOS prepared by the manufacturer enables the wifi and Bluetooth but I see that in yocto kernel the driver is not available. First, I tried by using menuconfig on virtual/ kernel but couldn't find the option for the particular  driver.

Secondly, I then did a 2step process by first running menuconfug and then adding rtl8723bs to kconfig. Then, compiling the driver threw errors in the source code of the driver in 4.4 kernel.

Thirdly, I did menuconfig in kernel , replaced the driver source in kernel source in my build with latest (found from github wiki) and built the kernel successfullly. 

However, after the board boots, dmesg shows the kernel loading rtl8723bs successfully but wlan0 was not found and errors like ' wifi has not started successfully ' are printed here and there .

Can you add support for rtl8723bs chip for Tinkerboard as I seem to be mixing things up to make it work? If the support cannot be added as it is an out of tree module , please guide me to get it working. Thnx

PS: the USB based wifi dongles work ok on the board but I neeeded to copy the rt2800.bin files manually to the board. The USB wifi solution won't work for me practically as I am using a battery to run Tinkerboard and I prefer the onboard chip due to its tighter integration to the board (also USB wifi is clunky in appearance and performance)

wzyy2, 2017-03-27:
https://github.com/rockchip-linux/kernel/tree/release-20160818-miniarm

wzyy2, 2017-03-27:
WIFI driver for tinker board is ready in this branch, you can cherry-pick them to release-4.4

sat8, 2017-03-27:
Gr8. Compiling with 'release-20160818-miniarm' branch resolved the issue. thnx

closed 2017-03-27 by sat8

teacupx, 2018-01-24:
Sorry to revive this old issue, but the 'release-20160818-miniarm' is not available anymore. Any chance to get WiFi working in the Tinkerboard now?

wzyy2, 2018-01-25:
It's better to use tinkerboard kernel.
You can change kernel_src_uri to https://github.com/TinkerBoard/debian_kernel, kernel config to `miniarm_rk3288_defconfig`

teacupx, 2018-01-25:
Are you sure this will enable wifi? In Debian image, wifi doesn't work either.

wzyy2, 2018-01-25:
`rockchip-linux/release-4.4` is the upstream branch of `TinkerBoard/debian_kernel`. 
ASUS guys have push their changes, like wifi driver, to `TinkerBoard/debian_kernel`, so if you want to get full features for TKB, you should use that kernel, or cherry-pick commits from `TinkerBoard/debian_kernel` to `rockchip-linux/release-4.4`.

Oh, you also have to put wifi firmware to rootfs.