rockchip-linux/meta-rockchip #6: RK3288 - IMX219 CSI sensor compatibility

sat8, 2017-03-27:
Hi 

I am trying to find a decent camera sensor that can go onto Tinker board. I am looking at some of the off the shelf components that I can get hold of in the UK and raspberry pi's 8 MP sensor https://www.raspberrypi.org/products/camera-module-v2/ fits my requirement. 

Can you please advice if there are any incompatibilities with the sensor module? I am guessing that the board has a MIPI CSI interface and as such this MIPI CSI camera sensor should be ok.

Supposing there is an incompatibility would you recommend using a FireFly RK3288 camera?
https://www.amazon.co.uk/Firefly-RK3288-OV13850-Camera-Module/dp/B014B6A256 

Looking through the kernel source, a bunch of i2c drivers are listed in https://github.com/rockchip-linux/kernel/tree/69db1b530898a8bfc559bd8035a95c20d54dcdb9/drivers/media/i2c/soc_camera . Is it possible any of these CSI MIPI cameras will work with Tinker board?

Thnx

wzyy2, 2017-03-27:
Linux v4l2 CSI driver aren't completely done, I recommend not to use.....

If you want to try it, then you can use IMX219 (same with rpi) in Tinker board.
https://github.com/rockchip-linux/kernel/commit/dad1c7268399f558f9c6324609b626d8bf15c057

sat8, 2017-03-27:
Gr8. I will try this and feedback what happens. Should I be using gstreamer's v4l2src to fetch raw data? If so what is the expected format (eg: video/x-raw etc.,). I also see that there is a to-camera repo that demonstrates camera but it feels more like android app. Pls confirm a way to test the camera.

wzyy2, 2017-03-28:
gstreamer plugin and compatible driver are still in develop.

> I also see that there is a to-camera repo that demonstrates camera but it feels more like android app.
We only have this app which was ported from android to test the camera now.
So I recommend not to use it......

closed 2017-03-28 by wzyy2

kozuch, 2017-07-16:
@wzyy2 Any news on this topic? I am looking for a low level camera interface like RPi's MMAL but with gigabit ethernet - will Tinker Board ever be able to give similar low level access to some of its supported camera modules? Is any camera actually working today?

wzyy2, 2017-07-16:
https://github.com/rockchip-linux/rk-rootfs-build/blob/master/overlay-debug/usr/local/bin/test_camera.sh

We are making that driver support videobuf2, so it can be used in gstreamer with v4l2src, though
it just bypass images, ISP process are not available.

It will be updated soon.

proxax, 2017-09-28:
Hello

Is/has anyone changed the exposure and resolution settings for a USB web camera? The camera application crashes as soon as the settings are updated.

We're using a Logitech C920 1080p Web Camera. We've tried the following three tools/config apps; 
- guvcview
- uvcdynctrl
- v4l2ucp

**Would appreciate if anyone can share their experience and steps they took to resolve the same.**

mohamadnasr, 2018-03-06:
hello 
i use tinker os v2.4 with IMX219 camera(raspi cam v2) 
when i uses Gstremer it show just Green channel

what can id do?
the problem is in the Driver? 
please help me

wzyy2, 2018-03-07:
You can ask in here.
https://www.tinkerboarding.co.uk/forum