rockchip-linux/meta-rockchip #7: yocto issues with meta-rockchip MACHINE tinker-rk3288

tx0h, 2017-03-27:
hi,

after configuring bblayers.conf and local.conf i started bitbake like here:

```
ugly:~/YOCTO/build$ bitbake core-image-minimal
WARNING: Host distribution "debian-9" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Parsing recipes: 100% |########################################################################################################################################################| Time: 0:00:36
Parsing of 865 .bb files complete (0 cached, 865 parsed). 1334 targets, 80 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'virtual/kernel'
ERROR: linux-yocto-rt PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-yocto PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-yocto-tiny PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-yocto-tiny PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-rockchip PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-yocto-rt PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-rockchip PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-yocto PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-yocto-rt PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-yocto-tiny PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-yocto-rt PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-yocto-tiny PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-rockchip PROVIDES virtual/kernel but was skipped: Set PREFERRED_PROVIDER_virtual/kernel to linux-rockchip to enable it
ERROR: linux-veyron PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-yocto PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-dummy PROVIDES virtual/kernel but was skipped: PREFERRED_PROVIDER_virtual/kernel set to linux, not linux-dummy
ERROR: linux-yocto PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-yocto-rt PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-yocto-tiny PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-yocto PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: linux-yocto-dev PROVIDES virtual/kernel but was skipped: incompatible with machine tinker-rk3288 (not in COMPATIBLE_MACHINE)
ERROR: Required build target 'core-image-minimal' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-minimal', 'virtual/kernel']

Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

then i added tinker-rk3288 to the linux_4.10.bb in meta-rockchip and bitbake it again:

```
iNITIAlising tasks: 100% |#####################################################################################################################################################| Time: 0:00:03
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: linux-4.10-rc8-r1 do_compile: oe_runmake failed
ERROR: linux-4.10-rc8-r1 do_compile: Function failed: do_compile (log file is located at /home/beeker23/YOCTO/build/tmp/work/tinker_rk3288-poky-linux-gnueabi/linux/4.10-rc8-r1/temp/log.do_compile.6768)
ERROR: Logfile of failure stored in: /home/beeker23/YOCTO/build/tmp/work/tinker_rk3288-poky-linux-gnueabi/linux/4.10-rc8-r1/temp/log.do_compile.6768
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8 HOSTCC=gcc  HOSTCPP=gcc  -E zImage CC=arm-poky-linux-gnueabi-gcc  -mno-thumb-interwork -marm -fuse-ld=bfd LD=arm-poky-linux-gnueabi-ld.bfd
| NOTE: make -j 8 HOSTCC=gcc  HOSTCPP=gcc  -E rk3288-miniarm.dtb
| ERROR: oe_runmake failed
|   CHK     include/config/kernel.release
|   GEN     ./Makefile
|   CHK     include/generated/uapi/linux/version.h
|   Using /home/beeker23/YOCTO/build/tmp/work-shared/tinker-rk3288/kernel-source as source for kernel
|   CHK     include/generated/utsrelease.h
|   CC      scripts/mod/empty.o
|   CC      scripts/mod/devicetable-offsets.s
|   MKELF   scripts/mod/elfconfig.h
|   HOSTCC  scripts/mod/modpost.o
|   HOSTCC  scripts/mod/sumversion.o
|   GEN     scripts/mod/devicetable-offsets.h
|   HOSTCC  scripts/mod/file2alias.o
|   HOSTLD  scripts/mod/modpost
|   CHK     include/generated/timeconst.h
|   CHK     include/generated/bounds.h
|   CHK     include/generated/asm-offsets.h
|   CALL    /home/beeker23/YOCTO/build/tmp/work-shared/tinker-rk3288/kernel-source/scripts/checksyscalls.sh
|   CHK     include/generated/compile.h
|   UPD     include/generated/compile.h
|   CC      init/version.o
|   LD      init/built-in.o
|   GEN     .version
|   CHK     include/generated/compile.h
|   UPD     include/generated/compile.h
|   CC      init/version.o
|   LD      init/built-in.o
|   LD      vmlinux.o
|   MODPOST vmlinux.o
|   KSYM    .tmp_kallsyms1.o
|   KSYM    .tmp_kallsyms2.o
|   LD      vmlinux
|   SORTEX  vmlinux
|   SYSMAP  System.map
|   OBJCOPY arch/arm/boot/Image
|   Kernel: arch/arm/boot/Image is ready
|   GZIP    arch/arm/boot/compressed/piggy_data
|   AS      arch/arm/boot/compressed/piggy.o
|   LD      arch/arm/boot/compressed/vmlinux
|   OBJCOPY arch/arm/boot/zImage
|   Kernel: arch/arm/boot/zImage is ready
|   CC      scripts/mod/empty.o
|   CC      scripts/mod/devicetable-offsets.s
|   GEN     scripts/mod/devicetable-offsets.h
|   MKELF   scripts/mod/elfconfig.h
|   HOSTCC  scripts/mod/sumversion.o
|   HOSTCC  scripts/mod/file2alias.o
|   HOSTCC  scripts/mod/modpost.o
|   HOSTLD  scripts/mod/modpost
| make[3]: *** No rule to make target 'arch/arm/boot/dts/rk3288-miniarm.dtb'.  Stop.
| arch/arm/Makefile:340: recipe for target 'rk3288-miniarm.dtb' failed
| make[2]: *** [rk3288-miniarm.dtb] Error 2
| Makefile:150: recipe for target 'sub-make' failed
| make[1]: *** [sub-make] Error 2
| Makefile:24: recipe for target '__sub-make' failed
| make: *** [__sub-make] Error 2
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/beeker23/YOCTO/build/tmp/work/tinker_rk3288-poky-linux-gnueabi/linux/4.10-rc8-r1/temp/log.do_compile.6768)
ERROR: Task (/home/beeker23/YOCTO/meta-rockchip/recipes-kernel/linux/linux_4.10.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1541 tasks of which 1480 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/beeker23/YOCTO/meta-rockchip/recipes-kernel/linux/linux_4.10.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
beeker23@ugly:~/YOCTO/build$ 
```

which means, there is no receipt for rk3288-miniarm.

now what?

i'm using git MASTER of yocto and git MASTER of meta-rockchip.

thanks,
tx0h

wzyy2, 2017-03-28:
http://opensource.rock-chips.com/wiki_Yocto

closed 2017-03-28 by wzyy2

tx0h, 2017-03-29:
hello jacob,

thank you. i followed your yocto wiki page,
but i still have some problems and would
be pleased if you could help me with some
hints in regards to the tinker board:

in another issue you pointed to another kernel
(release-20160818-miniarm) to cherry-pick the
wireless driver. i tried that with no luck.
whatdo i need to take over?
the whole rockchip_wlan directory or just the
rtl8723bs directory?

the original image (tinkerOS) is able to detect
the screen resolution and set the parameters
correctly for the console and for X.
rockchip-linux needs kernel parameter for
my 1024x600 hdmi for the console but X
didn't work on the 1024x600 screen, yet.

greetings,
thomax

wzyy2, 2017-03-30:
You could use this patch from libre guys.

https://github.com/LongChair/LibreELEC.tv/blob/pmp-tinker/projects/TinkerBoard/patches/linux/rockchip-4.4/linux-0003-wireless-update-rtl8723bs-driver.patch

About screen, could you do a experiment, replace kernel in tinker image 
with rockchip-linux kernel ?

tx0h wrote on 2017年03月30日 04:59: