rockchip-linux/meta-rockchip #8: kernel build fails: linux-rockchip_4.4.bb

wdmegrv, 2017-04-13:
``bitbake linux-rockchip``

The kernel build fails with the following error:
`| /home/egorov/yocto-builds-rk/build/tmp/work-shared/firefly-rk3288/kernel-source/scripts/extract-cert.c:21:25: fatal error: openssl/bio.h: No such file or directory
|  #include <openssl/bio.h>
|                          ^
| compilation terminated.
`
I think your kernel recipe should depend on openssl-native since you are using CONFIG_SYSTEM_TRUSTED_KEYRING which includes openssl header.

But a `DEPENDS += "openssl-native" ` in the recipe does not fix this problem. I think yocto messed up the compiler flags for the HOST tools.
Do you have a solution for this issue besides installing ssl-dev on my host machine?

wzyy2, 2017-04-13:
why not install ssl-dev on your host machine? - -

wdmegrv, 2017-04-13:
Installing it on the host machine feels very wrong. It should be fixed in yocto. This is also not the right approach. yocto gives u a way to handle such cases with *-native packages.
Also I don't want to install it on a build server.

wzyy2, 2017-04-14:
I'm not sure how to fix it.
I think you should ask it in yocto mailing list.

closed 2017-05-11 by wzyy2

amery, 2017-10-28:
appending `DEPENDS += "openssl-native"` to the .bb solves it for me...

wzyy2, 2017-11-02:
Thanks, i have add them.
https://github.com/rockchip-linux/meta-rockchip/commit/d353706d7e27ff43d7b444bb5e72da50fab019ba

samsonluk, 2018-05-20:
Problem still exit...

bitbake rk-image-multimedia

HOSTCC  scripts/extract-cert
| /usr/src/rk-yocto-bsp/build/tmp/work-shared/tinker-rk3288/kernel-source/scripts/extract-cert.c:21:25: fatal error: openssl/bio.h: No such file or directory
|  #include <openssl/bio.h>
|                          ^
| compilation terminated.
| make[3]: *** [scripts/extract-cert] Error 1
| make[3]: *** Waiting for unfinished jobs....
| make[2]: *** [scripts] Error 2
| make[1]: *** [sub-make] Error 2
| make: *** [__sub-make] Error 2
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /usr/src/rk-yocto-bsp/build/tmp/work/tinker_rk3288-rk-linux-gnueabi/linux-rockchip/4.4.103-r1/temp/log.do_compile.9691)
ERROR: Task (/usr/src/rk-yocto-bsp/sources/meta-rockchip/recipes-kernel/linux/linux-rockchip_4.4.bb:do_compile) failed with exit code '1'