rockchip-linux/meta-rockchip #11: MAC Address of eth0 is random on every boot

sat8, 2017-04-21:
Hi

I have run in to an issue with MAC address of the ethernet port getting a random address on every boot.
While looking for possible solutions on the web, a post  on raspberry pi forums indicated that the mac address is linked to cpu serial and the CPU serial being null can be a reason for kernel to assign random mac address. 

On Tinkerboard, the output of command **cat /proc/cpuinfo** returns a null serial for cpus.

[image]

Please advise a solution.

wzyy2, 2017-04-21:
Hi,

Tinker board usually read mac adress from i2c.
You can pick this patch to u-boot.
https://patchwork.ozlabs.org/patch/752938/

And if you update kernel to latest release-4.4, it will also use a fixed mac adress.

sat8, 2017-04-22:
great. This has fixed it. many thanks.

closed 2017-04-22 by sat8