rockchip-linux/meta-rockchip #15: gcc-runtime_7.1.bb:do_compile ERROR

ArneKorsika, 2017-07-01:
I followed the instructions on http://opensource.rock-chips.com/wiki_Yocto. When I use bitbake to build ``` rk-image-base ``` I was getting all sort of different errors ranging from openjade to different warnings of not being able to fetch certain libraries. However, since the last repo sync from 28.6.2017 I'm getting a constant error. Which is happening in the ```gcc/gcc-runtime_7.1.bb:do_compile```. 

My host system is **Ubuntu 16.04 LTS**

The last few lines of the output:
```
| .libs/compatibility-c++0x.o: In function `__static_initialization_and_destruction_0':
| /usr/src/debug/gcc-runtime/7.1.0-r0/gcc-7.1.0/build.arm-rk-linux-gnueabi.arm-rk-linux-gnueabi/arm-rk-linux-gnueabi/libstdc++-v3/src/../../../../../../../../../work-shared/gcc-7.1.0-r0/gcc-7.1.0/libstdc++-v3/src/c++11/compatibility-c++0x.cc:215: undefined reference to `__dso_handle'
| ../src/c++11/.libs/libc++11convenience.a(system_error.o): In function `__static_initialization_and_destruction_0':
| /usr/src/debug/gcc-runtime/7.1.0-r0/gcc-7.1.0/build.arm-rk-linux-gnueabi.arm-rk-linux-gnueabi/arm-rk-linux-gnueabi/libstdc++-v3/src/c++11/../../../../../../../../../../work-shared/gcc-7.1.0-r0/gcc-7.1.0/libstdc++-v3/src/c++11/system_error.cc:71: undefined reference to `__dso_handle'
| ../src/c++98/.libs/libc++98convenience.a(mt_allocator.o): In function `get_freelist':
| /usr/src/debug/gcc-runtime/7.1.0-r0/gcc-7.1.0/build.arm-rk-linux-gnueabi.arm-rk-linux-gnueabi/arm-rk-linux-gnueabi/libstdc++-v3/src/c++98/../../../../../../../../../../work-shared/gcc-7.1.0-r0/gcc-7.1.0/libstdc++-v3/src/c++98/mt_allocator.cc:64: undefined reference to `__dso_handle'
| ../src/c++98/.libs/libc++98convenience.a(c++locale.o): In function `std::get_catalogs()':
| /usr/src/debug/gcc-runtime/7.1.0-r0/gcc-7.1.0/build.arm-rk-linux-gnueabi.arm-rk-linux-gnueabi/arm-rk-linux-gnueabi/libstdc++-v3/src/c++98/c++locale.cc:251: undefined reference to `__dso_handle'
| ../src/c++11/.libs/libc++11convenience.a(condition_variable.o): In function `key_init':
| /usr/src/debug/gcc-runtime/7.1.0-r0/gcc-7.1.0/build.arm-rk-linux-gnueabi.arm-rk-linux-gnueabi/arm-rk-linux-gnueabi/libstdc++-v3/src/c++11/../../../../../../../../../../work-shared/gcc-7.1.0-r0/gcc-7.1.0/libstdc++-v3/src/c++11/condition_variable.cc:133: undefined reference to `__dso_handle'
| ../src/c++11/.libs/libc++11convenience.a(future.o):/usr/src/debug/gcc-runtime/7.1.0-r0/gcc-7.1.0/build.arm-rk-linux-gnueabi.arm-rk-linux-gnueabi/arm-rk-linux-gnueabi/libstdc++-v3/src/c++11/../../../../../../../../../../work-shared/gcc-7.1.0-r0/gcc-7.1.0/libstdc++-v3/src/c++11/future.cc:65: more undefined references to `__dso_handle' follow
| collect2: error: ld returned 1 exit status
| Makefile:609: recipe for target 'libstdc++.la' failed
| make[3]: *** [libstdc++.la] Error 1
| make[3]: Leaving directory '/media/arnekorsika/UbuntuMainFS/UbuntuStuff/Sudo/sudo-os/embeded-linux-evb/rk-yocto-bsp/test-bld/tmp/work/cortexa17hf-neon-vfpv4-rk-linux-gnueabi/gcc-runtime/7.1.0-r0/gcc-7.1.0/build.arm-rk-linux-gnueabi.arm-rk-linux-gnueabi/arm-rk-linux-gnueabi/libstdc++-v3/src'
| Makefile:641: recipe for target 'all-recursive' failed
| make[2]: *** [all-recursive] Error 1
| make[2]: Leaving directory '/media/arnekorsika/UbuntuMainFS/UbuntuStuff/Sudo/sudo-os/embeded-linux-evb/rk-yocto-bsp/test-bld/tmp/work/cortexa17hf-neon-vfpv4-rk-linux-gnueabi/gcc-runtime/7.1.0-r0/gcc-7.1.0/build.arm-rk-linux-gnueabi.arm-rk-linux-gnueabi/arm-rk-linux-gnueabi/libstdc++-v3/src'
| Makefile:510: recipe for target 'all-recursive' failed
| make[1]: *** [all-recursive] Error 1
| make[1]: Leaving directory '/media/arnekorsika/UbuntuMainFS/UbuntuStuff/Sudo/sudo-os/embeded-linux-evb/rk-yocto-bsp/test-bld/tmp/work/cortexa17hf-neon-vfpv4-rk-linux-gnueabi/gcc-runtime/7.1.0-r0/gcc-7.1.0/build.arm-rk-linux-gnueabi.arm-rk-linux-gnueabi/arm-rk-linux-gnueabi/libstdc++-v3'
| Makefile:417: recipe for target 'all' failed
| make: *** [all] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /media/arnekorsika/UbuntuMainFS/UbuntuStuff/Sudo/sudo-os/embeded-linux-evb/rk-yocto-bsp/test-bld/tmp/work/cortexa17hf-neon-vfpv4-rk-linux-gnueabi/gcc-runtime/7.1.0-r0/temp/log.do_compile.7652)
ERROR: Task (/media/arnekorsika/UbuntuMainFS/UbuntuStuff/Sudo/sudo-os/embeded-linux-evb/rk-yocto-bsp/sources/poky/meta/recipes-devtools/gcc/gcc-runtime_7.1.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1730 tasks of which 600 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /media/arnekorsika/UbuntuMainFS/UbuntuStuff/Sudo/sudo-os/embeded-linux-evb/rk-yocto-bsp/sources/poky/meta/recipes-devtools/gcc/gcc-runtime_7.1.bb:do_compile
Summary: There were 6 WARNING messages shown.
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.
```

wzyy2, 2017-07-03:
Could you use docker to install a clean ubuntu 16.04?

ArneKorsika, 2017-07-08:
Before I started with Docker I gave it another try on my newly installed Ubunutu 16.04.2 LTS. I synced my working directory with ```repo sync``` and ran ```bitbake``` and in the process I got an error:
```ERROR: linux-rockchip-4.4.70-r1 do_compile: oe_runmake failed ERROR: linux-rockchip-4.4.70-r1 do_compile: Function failed: do_compile```

And then as you recommended I've tried the same in Docker with a clean install of Ubuntu 16.04.2 LTS and got exactly the same error. 

So my process was as follows:
1. On the host (*Ubuntu 16.04.2 LTS*) I created an empty directory `rk-yocto-work-dir` in the ```/home``` folder;
2. In `rk-yocto-work-dir` dir I ran:
2.1 ```repo init --repo-url=https://github.com/rockchip-linux/repo -u https://github.com/rockchip-linux/manifests -b yocto```;
2.2 ```repo sync```;
3. Ran ```sudo apt-get install gawk wget git-core diffstat unzip texinfo build-essential chrpath socat cpio python python3 pip3 pexpect libsdl1.2-dev xterm make xsltproc docbook-utils fop dblatex xmlto python-git```;
*By the way there is a typo in the wiki for this command. 
This ```python3 pip3 pexpect``` should be ```python3-pip python3-pexpect```. So the command should be ```sudo apt-get install gawk wget git-core diffstat unzip texinfo  build-essential chrpath socat cpio python python3-pip python3-pexpect libsdl1.2-dev xterm make xsltproc docbook-utils fop dblatex xmlto python-git```*;
4. In the working directory I created a dir ```mkdir test-build```;
5. Ran ```MACHINE=evb-rk3288 DISTRO=rk-x11 . ./setup-environment -b test-build```;
6. Ran ```bitbake core-image-base```.
*After some time at around 57% of all the tasks executed I get an ERROR ```ERROR: linux-rockchip-4.4.70-r1 do_compile: oe_runmake failed ERROR: linux-rockchip-4.4.70-r1 do_compile: Function failed: do_compile```*.

Am I missing anything?

wzyy2, 2017-07-10:
try this? https://github.com/rockchip-linux/docker-rockchip

ArneKorsika, 2017-07-10:
Last night I found what the problem was. The ```libssl-dev``` was missing. The thing is that yocto/oe doesn't let you know that it's missing in a straight forward way. So I fixed it by installing it...```sudo apt-get install libssl-dev```
It seems libssl-dev lib doesn't come with Ubuntu 16.04 LTS out of the box. Should this lib be added in the Build Environment essentials?

wzyy2, 2017-07-10:
Thx
i have add it to wiki.

closed 2017-07-10 by wzyy2

ArneKorsika, 2017-07-10:
Also it would be nice to add a fix wiki. I have described it in the second comment under point 3.