rockchip-linux/meta-rockchip #16: build error about weston 2.0.1

badtoperfect, 2017-08-02:
Yocto Build Error:
weston-2.0.1-r0 do_fetch: Fetcher failure: Unable to find revision ba89ba384c88869fe6c6d6c669ed8b067c681c5f in branch wip/2017-04/atomic-v11-WIP even from upstream

wzyy2, 2017-08-02:
Thx for info.

It's a personal experimental branch, we use it to demo weston's new feature. 
Since it was removed, i will switch it back to the default upstream weston.

We have not completely supported weston in this bsp. The upstream weston have not merge some features and the performance are limited.

wzyy2, 2017-08-02:
removed.

https://github.com/rockchip-linux/meta-rockchip/commit/e25a953de35e26ceb73b9107ac802900e2a0b3b1
https://github.com/rockchip-linux/meta-rockchip-extra/commit/ac365fe386daf6ca557b278edb2b8ab6c6e366b2

closed 2017-08-02 by wzyy2