rockchip-linux/meta-rockchip #17: why not the firefly-rk3399 machine use firefly-rk3399_defconfig(UBOOT_MACHINE) ?

badtoperfect, 2017-08-16:
when I build the u-boot-rockchip for firefly-rk3399 machine ,there is a build error as below,
but when I change the UBOOT_MACHINE to firefly-rk3399_defconfig ,there has no error for u-boot-rockchip

**Error:**

/home/yangzhentao/yocto/rockchip/rk-yocto-bsp/build/tmp/work/firefly_rk3399-rk-linux/u-boot-rockchip/v2017.05+gitAUTOINC+9fe6f2383c-r0/git/scripts/Makefile.spl:359: recipe for target 'checkdtoc' failed
| make[2]: *** [checkdtoc] Error 1
| make[2]: *** Waiting for unfinished jobs....
| /home/yangzhentao/yocto/rockchip/rk-yocto-bsp/build/tmp/work/firefly_rk3399-rk-linux/u-boot-rockchip/v2017.05+gitAUTOINC+9fe6f2383c-r0/git/Makefile:1396: recipe for target 'spl/u-boot-spl' failed
| make[1]: *** [spl/u-boot-spl] Error 2
| make[1]: Leaving directory '/home/yangzhentao/yocto/rockchip/rk-yocto-bsp/build/tmp/work/firefly_rk3399-rk-linux/u-boot-rockchip/v2017.05+gitAUTOINC+9fe6f2383c-r0/build'
| Makefile:150: recipe for target 'sub-make' failed
| make: *** [sub-make] Error 2
| make: Leaving directory '/home/yangzhentao/yocto/rockchip/rk-yocto-bsp/build/tmp/work/firefly_rk3399-rk-linux/u-boot-rockchip/v2017.05+gitAUTOINC+9fe6f2383c-r0/git'
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/yangzhentao/yocto/rockchip/rk-yocto-bsp/build/tmp/work/firefly_rk3399-rk-linux/u-boot-rockchip/v2017.05+gitAUTOINC+9fe6f2383c-r0/temp/log.do_compile.15448)
ERROR: Task (/home/yangzhentao/yocto/rockchip/rk-yocto-bsp/sources/meta-rockchip/recipes-bsp/u-boot/u-boot-rockchip_git.bb:do_compile) failed with exit code '

wzyy2, 2017-08-16:
It was just when i add firefly-rk3399 support, there is no firefly-rk3399_defconfig in u-boot.

It seems firefly-rk3399_defconfig have enable SPL support which make issues, you can simply disable it in firefly-rk3399_defconfig, it's not used in yocto.

badtoperfect, 2017-08-16:
Hmm....the error occurred when I used the "evb-rk3399_defconfig",not "firefly-rk3399_defconfig",
when I used "firefly-rk3399_defconfig",there is no error occurred.....

wzyy2, 2017-08-16:
- -OK...

wzyy2, 2017-08-16:
I will update it to use firefly-rk3399_defconfig.

closed 2017-08-16 by wzyy2