rockchip-linux/meta-rockchip #19: there is a error broken the build when I update to eecee11ec8e32a2e44dcef684fbcf9079939c345.

badtoperfect, 2017-08-31:
| checking whether the C compiler works... yes
| checking for C compiler default output file name... a.out
| checking for suffix of executables...
| checking whether we are cross compiling... configure: error: in `/home/yangzhentao/yocto/rockchip/yocto/build_systemd/tmp/work/aarch64-evo-linux/gstreamer1.0-rockchip-extra/20170829-r0/build':
| configure: error: cannot run C compiled programs.
| If you meant to cross compile, use `--host'.
| See `config.log' for more details
|   configure failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/yangzhentao/yocto/rockchip/yocto/build_systemd/tmp/work/aarch64-evo-linux/gstreamer1.0-rockchip-extra/20170829-r0/temp/log.do_configure.8673)
ERROR: Task (/home/yangzhentao/yocto/rockchip/yocto/sources/meta-rockchip/recipes-multimedia/gstreamer-rockchip/gstreamer1.0-rockchip-extra_20170829.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5419 tasks of which 5413 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/yangzhentao/yocto/rockchip/yocto/sources/meta-rockchip/recipes-multimedia/gstreamer-rockchip/gstreamer1.0-rockchip-extra_20170829.bb:do_configure
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

closed 2017-08-31 by badtoperfect

wzyy2, 2017-08-31:
My mistake
https://github.com/rockchip-linux/meta-rockchip/commit/5398a375772dfb1631487fc16bac251f299c526c
It will fix this problem

badtoperfect, 2017-08-31:
It works well now, Thank you ~