rockchip-linux/meta-rockchip #20: Build error with rockchip-mali

johnobe, 2017-09-08:
Hello,

I have an ASUS Tinkerboard (RK3288) and was building an image and encountered an error when building an image. To workaround the error, I commented out the 4 lines added in https://github.com/rockchip-linux/meta-rockchip/commit/4238da791d894a3da64af34fe21edc5767689906. I figured I didn't need those 4 lines on the RK3288.

```
DEBUG: sed -e 's:^[^/]*/:/home/john/yocto/rk-yocto-bsp/tinkerboard/tmp/work/cortexa17hf-neon-vfpv4-rk-linux-gnueabi/rockchip-mali/git-r0/recipe-sysroot-native/:g' /home/john/yocto/rk-yocto-bsp/tinkerboard/tmp/sysroots-components/x86_64/dpkg-native/fixmepath /home/john/yocto/rk-yocto-bsp/tinkerboard/tmp/sysroots-components/x86_64/perl-native/fixmepath /home/john/yocto/rk-yocto-bsp/tinkerboard/tmp/sysroots-components/x86_64/gettext-native/fixmepath | xargs sed -i -e 's:FIXMESTAGINGDIRTARGET:/home/john/yocto/rk-yocto-bsp/tinkerboard/tmp/work/cortexa17hf-neon-vfpv4-rk-linux-gnueabi/rockchip-mali/git-r0/recipe-sysroot:g; s:FIXMESTAGINGDIRHOST:/home/john/yocto/rk-yocto-bsp/tinkerboard/tmp/work/cortexa17hf-neon-vfpv4-rk-linux-gnueabi/rockchip-mali/git-r0/recipe-sysroot-native:g' -e 's:FIXME_COMPONENTS_DIR:/home/john/yocto/rk-yocto-bsp/tinkerboard/tmp/sysroots-components:g' -e 's:FIXME_HOSTTOOLS_DIR:/home/john/yocto/rk-yocto-bsp/tinkerboard/tmp/hosttools:g' -e 's:FIXME_PKGDATA_DIR:/home/john/yocto/rk-yocto-bsp/tinkerboard/tmp/pkgdata/tinker-rk3288:g'
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing python function do_package_write_deb
DEBUG: Executing python function read_subpackage_metadata
DEBUG: Python function read_subpackage_metadata finished
DEBUG: Executing python function do_package_deb
ERROR: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_package_deb(d)
     0003:
File: '/home/john/yocto/rk-yocto-bsp/sources/poky/meta/classes/package_deb.bbclass', lineno: 241, function: do_package_deb
     0237:        for key in list(rprovides.keys()):
     0238:            if key.startswith('/'):
     0239:                del rprovides[key]
     0240:        rprovides = collections.OrderedDict(sorted(rprovides.items(), key=lambda x: x[0]))
 *** 0241:        debian_cmp_remap(rprovides)
     0242:        rreplaces = bb.utils.explode_dep_versions2(localdata.getVar("RREPLACES") or "")
     0243:        debian_cmp_remap(rreplaces)
     0244:        rconflicts = bb.utils.explode_dep_versions2(localdata.getVar("RCONFLICTS") or "")
     0245:        debian_cmp_remap(rconflicts)
File: '/home/john/yocto/rk-yocto-bsp/sources/poky/meta/classes/package_deb.bbclass', lineno: 205, function: debian_cmp_remap
     0201:            #   '<' = less or equal
     0202:            #   '>' = greater or equal
     0203:            # adjust these to the '<<' and '>>' equivalents
     0204:            #
 *** 0205:            for dep in var:
     0206:                if '(' in dep:
     0207:                    newdep = dep.replace('(', '__')
     0208:                    newdep = newdep.replace(')', '__')
     0209:                    if newdep != dep:
Exception: RuntimeError: OrderedDict mutated during iteration

DEBUG: Python function do_package_deb finished
DEBUG: Python function do_package_write_deb finished
ERROR: Function failed: do_package_deb
```

Please let me know if you need any further information.

wzyy2, 2017-09-08:
It seems it will break build with old poky branch.
Let me think how to fix it.

johnobe, 2017-09-08:
Great. Thank you for looking at it.

wzyy2, 2017-09-08:
If you are using `pyro` branch, you could update `meta-rockchip` to fix it.
If you are using `master` branch, you should update `poky` to fix it, `repo sync poky`.

johnobe, 2017-09-08:
Thanks. I am using the `master` branch. I followed your advice and ran `repo poky sync` and was able to resolve the issue.

closed 2017-09-08 by johnobe