rockchip-linux/meta-rockchip #30: Problem with TRUST_IMG handling

ydirson, 2018-03-31:
context: I am trying to integrate a rk3399 build in a pre-existing image/distro, so I'm not using the repo-based stuff in this particular case, and copied relevant definitions from the rk image and distro definitions.

The build fails at final stage with:
```
+ dd if=/redacted/build/tmp/deploy/images/vs-rk3399/trust.img of=shadowdev-image-x11-vs-rk3399-gpt.img conv=notrunc,fsync seek=24576
dd: failed to open ‘/redacted/build/tmp/deploy/images/vs-rk3399/trust.img’: No such file or directory
```

Digging a bit, I notice that rockchip-gpt-img.bbclass TRUST_IMG = "rk-binary/trust.img", but that ${TRUST_IMG} expands to "trust.img", as can be seen in temp/run.do_image_rockchip_gpt_img:
```
        dd if=/redacted/build/tmp/deploy/images/vs-rk3399/trust.img of=shadowdev-image-x11-vs-rk3399-gpt.img conv=notrunc,fsync seek=${ATF_START}
```

If I rename the TRUST_IMG occurences in rockchip-gpt-img.bbclass to RKTRUST_IMG, then things work as expected, and dd is directed to rb-binary/trust.img as it should, and I can check that ${TRUST_IMG} still expands to plain "trust.img", but I can't find out who sets it to this value.

In the case of the standard build, something seems to copy ${DEPLOY_DIR_IMAGE}/rk-binary/trust.img to ${DEPLOY_DIR_IMAGE}/trust.img, although it's not clear to me at first sight where that copy is done.

wzyy2, 2018-03-31:
https://github.com/rockchip-linux/meta-rockchip/commit/226b2b3f4b584943cd1f0436cfae6285edbefe10

ydirson, 2018-03-31:
Does the job fine, thanks.
However, I'm on the rocko branch, so whatever caused the problem on poky/master was likely cherry-picked there too, you may want to apply it on your rock branch as well.

wzyy2, 2018-03-31:
@yanghanxing @Caesar-github 
Can you do that?

closed 2018-04-17 by hizukiayaka