rockchip-linux/meta-rockchip #32: Kernel panic booting "core-image-minimal"

pauldotknopf, 2018-04-17:
```
Build Configuration:
BB_VERSION           = "1.36.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "evb-rk3328"
DISTRO               = "poky"
DISTRO_VERSION       = "2.4.2"
TUNE_FEATURES        = "aarch64"
TARGET_FPU           = ""
meta
meta-poky
meta-yocto-bsp       = "rocko:fdeecc901196bbccd7c5b1ea4268a2cf56764a62"
meta                 = "rocko:1b18cdf6b8bdb00ff5df165b9ac7bc2b10c87d57"
meta-oe              = "rocko:dacfa2b1920e285531bec55cd2f08743390aaf57"
meta-rockchip        = "rocko:eae0225c335b4e5822dd7c144d5ed6219045e7b7"
meta-camera-source   = "master:4060627fac4c4b2e34d03b04ca94af17e78d13ef"
```

Here is my [image-gpt.img](https://drive.google.com/file/d/12-mT4wTUG9q4mm_uVO7TfVGIm6_aiYCl/view?usp=sharing) file if you want to try to boot it.

I don't have debug serial port or UART, so it is hard to get you a clean boot log, but I took some pictures. Maybe it helps. If not, I can try to hookup a boot debug port.

[image]
[image]
[image]
[image]
[image]

Here is a picture of my board (RK3328).

[image]
[image]

When I use the provided Etcher app and flash Debian, it boots fine. It must be something in the Yocto configuration.

closed 2018-04-17 by hizukiayaka

pauldotknopf, 2018-04-17:
Wrong repo?

wzyy2, 2018-04-18:
You'd better give a complete log