rockchip-linux/meta-rockchip #48: Support for NANOPC-T4

nandra, 2020-06-30:
I plan to buy https://www.friendlyarm.com/index.php?route=product/product&product_id=225 but looking at machines seems not yet supported by this meta layer. Does anyone work on it already? Otherwise I can also contribute. Thanks.

JeffyCN, 2020-07-15:
sorry, i'm not familiar(or be able to test) with this board.

adding support for new boards usually needs patches for kernel/u-boot(mostly config and dts), and a machine config(see examples under meta-rockchip/conf/machine)

nandra, 2020-07-15:
@JeffyCN ok thanks for info.

closed 2020-08-25 by JeffyCN

floion, 2022-01-10:
Hi, we have this https://github.com/balena-os/meta-nanopc-t4/tree/thud but we didn't update it a while though it should work