rockchip-linux/meta-rockchip #66: Chromium 95

MastaG, 2021-11-13:
Hi, would it be possible to rebase the v4l patches on top of Chromium 95.0.4638.69?

JeffyCN, 2021-11-13:
sorry, we are quit busy these days, i can try to do it early next month

rebasing the patches is not that hard, but there might be other issues during the upgrading

MastaG, 2021-11-13:
Thank you Jeffry,
I've tried to re-base for myself, but the code base is simply too complex for me to understand.

MastaG, 2021-11-28:
So I've attempted to rebase on top of Chromium 95 and 96: https://github.com/MastaG/odroid-oe-core/commit/4f36c895f2941356032bb29ad61468e49dbe65ce

It builds fine, however I can't test it atm because the latest XWayland update requires new symbols, without it I can't run the Gnome desktop.
See: https://github.com/rockchip-linux/libmali/issues/73

MastaG, 2021-11-29:
It seems Chromium doesn't run with --in-process-gpu any longer.
I've opened a bug report for further investigation: https://github.com/OSSystems/meta-browser/issues/591

closed 2021-12-08 by JeffyCN