rockchip-linux/meta-rockchip #67: Compatibility with newer Yocto releases

floion, 2022-01-10:
Hi,
I was looking to see if you support newer Yocto releases but I see the newest one is rocko. Do you plan to create a `honister` branch maybe for working with the Yocto Honister release?

JeffyCN, 2022-01-10:
rocko is no longer supported for a long time.

for hardknott: https://github.com/JeffyCN/yocto-manifests/tree/hardknott-next

honister is still WIP, it has a different override syntax than older versions, i'm planning to switch to it the next month:
https://github.com/JeffyCN/meta-rockchip/tree/honister
https://github.com/JeffyCN/yocto-conf/tree/honister

floion, 2022-01-12:
Hi, thanks for your reply. I looked in the layer.conf and saw that the approach you take is to use one branch that supports multiple Yocto releases. Are you taking into account creating branches of https://github.com/rockchip-linux/meta-rockchip/ for each Yocto release instead?

JeffyCN, 2022-01-12:
i'm planning to do it, but begin with honister

floion, 2022-01-18:
Great, thanks!

JeffyCN, 2022-01-27:
done

closed 2022-01-27 by JeffyCN

floion, 2022-02-15:
But I still do not see a Honister branch

JeffyCN, 2022-02-15:
> Thanks!

due to company policy, this github account(rockchip-linux) is no longer maintained.
check the newest version in:
https://github.com/JeffyCN/yocto-manifests/branches