rockchip-linux/meta-rockchip #68: Flashing wic generated image fails

nandra, 2022-01-12:
Hello, I'm using this layer and dunfell branch to build minimal image for 3399 excavator machine. I also fetched tools for updating and following readme I'm flashing like:
```
 sudo ./upgrade_tool wl 0 server/core-image-minimal-rockchip-rk3399.wic
[sudo] password for marek: 
Program Data in /home/marek/.config/upgrade_tool
Write LBA failed!
```

it flash until 70% then I get LBD write failed. Board boots but kernel hangs as t cannot mount rootfs. Any ideas? Thanks.

JeffyCN, 2022-01-12:
this kind of error usually related to usb cable.

you can try to re-flash it. if still failed, maybe re-flash the rootfs alone:
sudo upgrade_tool di -rootfs rootfs.img

nandra, 2022-01-17:
@JeffyCN yes sorry. When plugged cable directly to PC not to the USB extender flashing works. Closing.