rockchip-linux/meta-rockchip #69: Starting chromium on rk3399

nandra, 2022-01-17:
This is mor equestion then issue. I'm trying to start chromium on rk3399 but getting some issues. Chromium is compiled from dunfell branch and I've added to `IMAGE_FEATURES += "x11 x11-base`. When I `startx` I got:

```
[   630.005] (==) ModulePath set to "/usr/lib/xorg/modules"
[   630.005] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[   630.005] (II) Loader magic: 0x557886ecd0
[   630.005] (II) Module ABI versions:
[   630.005] 	X.Org ANSI C Emulation: 0.4
[   630.005] 	X.Org Video Driver: 24.1
[   630.005] 	X.Org XInput driver : 24.1
[   630.005] 	X.Org Server Extension : 10.0
[   630.008] (--) using VT number 2

[   630.008] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[   630.010] (II) xfree86: Adding drm device (/dev/dri/card0)
[   630.015] (II) no primary bus or device found
[   630.015] 	falling back to /sys/devices/platform/display-subsystem/drm/card0
[   630.015] (II) LoadModule: "glx"
[   630.016] (WW) Warning, couldn't open module glx
[   630.016] (EE) Failed to load module "glx" (module does not exist, 0)
[   630.016] (II) LoadModule: "modesetting"
[   630.016] (WW) Warning, couldn't open module modesetting
[   630.016] (EE) Failed to load module "modesetting" (module does not exist, 0)
[   630.016] (EE) No drivers available.
[   630.016] (EE) 
Fatal server error:
[   630.016] (EE) no screens found(EE) 
```

then chromium startup fails it cannot find display which is right. Any ideas how to start chromium? I don't need X it can be on top of wayland also.

JeffyCN, 2022-01-17:
hi, according to the log, it's lacking modesetting driver.

that layer is tested under this manifests:
https://github.com/JeffyCN/yocto-manifests

the demo config would be:
https://github.com/JeffyCN/yocto-conf/blob/master/include/display.conf#L10

it could support wayland(weston) and x11.

nandra, 2022-01-17:
@JeffyCN ok thanks. I treid with X11 but seems chromium cannot be started:
log can be found here: https://pastebin.com/sr7nK2BG

Maybe related to this: https://github.com/OSSystems/meta-browser/issues/588 ?

JeffyCN, 2022-01-17:
this function has been tested for a long time. please try the tested manifests and demo config.

JeffyCN, 2022-01-17:
this config is using weston by default:
https://github.com/JeffyCN/yocto-conf/blob/master/include/display.conf

to switch to x11, modify:
```
# For weston
DISTRO_FEATURES_remove += " x11"
DISTRO_FEATURES_append += " wayland"
IMAGE_INSTALL_append += " weston weston-init weston-examples"

# For x11
#DISTRO_FEATURES_remove += " wayland"
#DISTRO_FEATURES_append += " x11"
#IMAGE_FEATURES_append += " x11-base"
#IMAGE_INSTALL_append += " xf86-video-modesetting xserver-xorg-module-exa"
```

to
```
# For weston
#DISTRO_FEATURES_remove += " x11"
#DISTRO_FEATURES_append += " wayland"
#IMAGE_INSTALL_append += " weston weston-init weston-examples"

# For x11
DISTRO_FEATURES_remove += " wayland"
DISTRO_FEATURES_append += " x11"
IMAGE_FEATURES_append += " x11-base"
IMAGE_INSTALL_append += " xf86-video-modesetting xserver-xorg-module-exa"
```

nandra, 2022-01-21:
ok thanks for info.

closed 2022-01-21 by nandra