rockchip-linux/meta-rockchip #71: glmark2 run issue

nandra, 2022-01-25:
I'm trying to test gpu using glmark2 tool. When run on sapphire-excavator I'm getting:
``` glmark2 -d
Error: GLX version >= 1.3 is required
Error: Error: Couldn't get GL visual config!
Error: main: Could not initialize canvas
```
I checked and libGL from mesa is 1.2 only (I'm on dunfell release). Does anybody have some idea? I tried to use mesa from later yocto but it doesn't build. Thanks.

nandra, 2022-01-26:
Could be this related to: https://github.com/rockchip-linux/meta-rockchip/issues/69#issuecomment-1014323180

nandra, 2022-01-26:
OK glamrk2 I resolved by adding :
```
PACKAGECONFIG:pn-glmark2 = \
	"${@bb.utils.contains('DISTRO_FEATURES', 'x11 opengl', 'x11-gles2', '', d)} \
	${@bb.utils.contains('DISTRO_FEATURES', 'wayland opengl', 'wayland-gles2', '', d)} \
	drm-gles2"```

JeffyCN, 2022-01-27:
👍

closed 2022-01-27 by JeffyCN