rockchip-linux/meta-rockchip #72: chromium with wayland

nandra, 2022-01-27:
With my neverending tries to run chromium on rk3399 I use config from demo and was able to build chromium. But when I run it I get:
```
WAYLAND_DEBUG=1 DISPLAY=:0 chromium --no-sandbox
[375:375:0127/093754.320080:ERROR:power_monitor_device_source_stub.cc(11)] Not implemented reached in virtual bool base::PowerMonitorDeviceSource::IsOnBatteryPower()
[375:375:0127/093754.331052:ERROR:wayland_connection.cc(205)] Failed to connect to Wayland display
[375:375:0127/093754.331105:FATAL:ozone_platform_wayland.cc(231)] Failed to initialize Wayland platform
#0 0x00557a50a6b8 (/usr/lib/chromium/chromium-bin+0x798b6b7)
#1 0x00557a438380 (/usr/lib/chromium/chromium-bin+0x78b937f)

```

When board boots I see window manager and about command I run from serial shell. Does anybody know how to run chromium in wayland? Thanks.

JeffyCN, 2022-01-27:
please use the manifest in https://github.com/JeffyCN/yocto-manifests and follow the steps in readme.

then just run 'chromium'.

nandra, 2022-01-27:
Yes I did that but getting above still

JeffyCN, 2022-01-27:
that log means weston not started or XDG_RUNTIME_DIR env is wrong.

try to conform with pgrep weston and export XDG_RUNTIME_DIR=/run/user/0

nandra, 2022-01-27:
OK it move little forward:
```
root@rockchip-rk3399:~# export XDG_RUNTIME_DIR=/run/user/0
root@rockchip-rk3399:~# ps | grep weston
  257 root      4208 S    weston-launch -- --modules systemd-notify.so, --log=/var/log/weston.log
  266 root      411m S    /usr/bin/weston --modules systemd-notify.so, --log=/var/log/weston.log
  278 root     13460 S    /usr/libexec/weston-keyboard
  279 root     13004 S    /usr/libexec/weston-desktop-shell
  297 root      3192 S    grep weston
root@rockchip-rk3399:~# 
root@rockchip-rk3399:~# 
root@rockchip-rk3399:~# 
root@rockchip-rk3399:~# chromium 
[300:300:0127/104808.040799:ERROR:power_monitor_device_source_stub.cc(11)] Not implemented reached in virtual bool base::PowerMonitorDeviceSource::IsOnBatteryPower()
*** stack smashing detected ***: terminated
Received signal 6
#0 0x00555e6d06b8 (/usr/lib/chromium/chromium-bin+0x798b6b7)
#1 0x00555e5fe380 (/usr/lib/chromium/chromium-bin+0x78b937f)
#2 0x00555e6d0640 (/usr/lib/chromium/chromium-bin+0x798b63f)
#3 0x007fa3b316a0 ([vdso]+0x69f)
#4 0x007fa11fe560 gsignal
#5 0x007fa11ebdd4 abort
#6 0x007fa12362b4 (/lib/libc-2.31.so+0x6e2b3)
#7 0x007fa12a59d0 __fortify_fail
#8 0x007fa12a59a4 __stack_chk_fail
#9 0x00555e6ef358 (/usr/lib/chromium/chromium-bin+0x79aa357)
#10 0x00555e10c1a4 (/usr/lib/chromium/chromium-bin+0x73c71a3)
#11 0x00555e10d68c (/usr/lib/chromium/chromium-bin+0x73c868b)
#12 0x00555e10b594 (/usr/lib/chromium/chromium-bin+0x73c6593)
#13 0x00555e10b7dc (/usr/lib/chromium/chromium-bin+0x73c67db)
#14 0x00555a3c489c ChromeMain
#15 0x007fa11ec110 __libc_start_main
#16 0x00555a3c46b4 (/usr/lib/chromium/chromium-bin+0x367f6b3)
[end of stack trace]
*** stack smashing detected ***: terminated
Received signal 6
#0 0x00558cc006b8 (/usr/lib/chromium/chromium-bin+0x798b6b7)
#1 0x00558cb2e380 (/usr/lib/chromium/chromium-bin+0x78b937f)
#2 0x00558cc00640 (/usr/lib/chromium/chromium-bin+0x798b63f)
#3 0x007f7e1586a0 ([vdso]+0x69f)
#4 0x007f7b825560 gsignal
#5 0x007f7b812dd4 abort
#6 0x007f7b85d2b4 (/lib/libc-2.31.so+0x6e2b3)
#7 0x007f7b8cc9d0 __fortify_fail
#8 0x007f7b8cc9a4 __stack_chk_fail
#9 0x00558cc1f358 (/usr/lib/chromium/chromium-bin+0x79aa357)
#10 0x00558c63c1a4 (/usr/lib/chromium/chromium-bin+0x73c71a3)
#11 0x00558c63d68c (/usr/lib/chromium/chromium-bin+0x73c868b)
#12 0x00558c63b594 (/usr/lib/chromium/chromium-bin+0x73c6593)
#13 0x00558c63b7dc (/usr/lib/chromium/chromium-bin+0x73c67db)
#14 0x0055888f489c ChromeMain
#15 0x007f7b813110 __libc_start_main
#16 0x0055888f46b4 (/usr/lib/chromium/chromium-bin+0x367f6b3)
[end of stack trace]
[300:343:0127/104808.471643:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[300:343:0127/104808.471854:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[300:300:0127/104808.713692:ERROR:cursor_loader.cc(114)] Failed to load a platform cursor of type kNull
[300:300:0127/104808.875709:ERROR:wayland_window.cc(394)] Not implemented reached in virtual void ui::WaylandWindow::SetWindowIcons(const gfx::ImageSkia &, const gfx::ImageSkia &)
*** stack smashing detected ***: terminated
Received signal 6
#0 0x00558cc006b8 (/usr/lib/chromium/chromium-bin+0x798b6b7)
#1 0x00558cb2e380 (/usr/lib/chromium/chromium-bin+0x78b937f)
#2 0x00558cc00640 (/usr/lib/chromium/chromium-bin+0x798b63f)
#3 0x007f7e1586a0 ([vdso]+0x69f)
#4 0x007f7b825560 gsignal
#5 0x007f7b812dd4 abort
#6 0x007f7b85d2b4 (/lib/libc-2.31.so+0x6e2b3)
#7 0x007f7b8cc9d0 __fortify_fail
#8 0x007f7b8cc9a4 __stack_chk_fail
#9 0x00558cc1f358 (/usr/lib/chromium/chromium-bin+0x79aa357)
[... 243 lines cut ...]
#3 0x007f94ec76a0 ([vdso]+0x69f)
#4 0x007f92594560 gsignal
#5 0x007f92581dd4 abort
#6 0x005582f1269c (/usr/lib/chromium/chromium-bin+0x798a69b)
#7 0x005582e592d0 (/usr/lib/chromium/chromium-bin+0x78d12cf)
#8 0x005582e59a48 (/usr/lib/chromium/chromium-bin+0x78d1a47)
#9 0x005582e59a8c (/usr/lib/chromium/chromium-bin+0x78d1a8b)
#10 0x005582f2f578 (/usr/lib/chromium/chromium-bin+0x79a7577)
#11 0x005580823d50 (/usr/lib/chromium/chromium-bin+0x529bd4f)
#12 0x0055801f7c4c (/usr/lib/chromium/chromium-bin+0x4c6fc4b)
#13 0x005582eb15c8 (/usr/lib/chromium/chromium-bin+0x79295c7)
#14 0x005582edf79c (/usr/lib/chromium/chromium-bin+0x795779b)
#15 0x005582edeed8 (/usr/lib/chromium/chromium-bin+0x7956ed7)
#16 0x005582f27aec (/usr/lib/chromium/chromium-bin+0x799faeb)
#17 0x005582ede860 (/usr/lib/chromium/chromium-bin+0x795685f)
#18 0x005582eef280 (/usr/lib/chromium/chromium-bin+0x796727f)
#19 0x005582eeeedc (/usr/lib/chromium/chromium-bin+0x7966edb)
#20 0x005582f285f8 (/usr/lib/chromium/chromium-bin+0x79a05f7)
#21 0x007f94e5d3e0 (/lib/libpthread-2.31.so+0x83df)
#22 0x007f9262cd5c (/lib/libc-2.31.so+0xced5b)
[end of stack trace]
Aborted
root@rockchip-rk3399:~# [333:336:0127/104809.440263:ERROR:ssl_client_socket_impl.cc(983)] handshake failed; returned -1, SSL error code 1, net_error -3
[333:336:0127/104809.443736:ERROR:ssl_client_socket_impl.cc(983)] handshake failed; returned -1, SSL error code 1, net_error -3
```

JeffyCN, 2022-01-27:
looks like memory corruption.

try release branch of manifest or newest(master)

nandra, 2022-01-27:
You mean yocto-next branch for meta-rockchip?

JeffyCN, 2022-01-27:
> You mean yocto-next branch for meta-rockchip?

the release branches were well tested. and i'm currently using master branch:
https://github.com/JeffyCN/yocto-manifests/branches

nandra, 2022-01-31:
@JeffyCN chromium works. Thanks.

closed 2022-01-31 by nandra